thanhtam-h/soem-w5500-rpi #2: Do you have any w5500 shield board for rpb 3+?

zhouq0725, 2018-06-07:
Will you provide w5500 board like [PIGATOR Ethernet](http://busware.de/tiki-index.php?page=PIG_ETH)?

thanhtam-h, 2018-06-07:
Yes, here it is. It works for any rpi version. I will write hardware diy soon
With **rpi1**:
[image]

With **rpi0 (and 0-W)**:
[image]

And **rpi2,3 (and 3b+)**:
[image]

zhouq0725, 2018-06-08:
Great. How can i buy it?

thanhtam-h, 2018-06-09:
You may follow at website www.simplerobot.net

closed 2018-08-29 by thanhtam-h