thanhtam-h/soem-w5500-rpi #3: SOEM1.3.0？

frankgp, 2018-08-06:
I found the latest version of SOEM is 1.3.3. ( 1.3.1 -> 1.3.3)
Do you have any resean?  Using SOEM 1.3.0?

thanhtam-h, 2018-08-06:
I started with 1.3.0 for STM32F407 few years ago and later found that 1.3.1 didn't provide much improvement of EtherCAT performance but  more platforms support (you may check changelog here https://github.com/OpenEtherCATsociety/SOEM/blob/master/ChangeLog)
For the latest added feature on 1.3.3, 'overlapping IOmap', I didn't try but will consider it.
Btw, did you build your own an EtherCAT master?

frankgp, 2018-08-06:
Thank you for your explain!

Yes! Now I try to make a EtherCAT master for 2 platforms, one is STM32F407 for small RTOS, the other is  imx6ul.
for imx6ul, the linux+xenomai, I using IGH ethermaster, I think it is better than SOEM.  but it cannot portting to MCU, so now I try to using SOEM for both platform.  

For STM32F407, I need 2 Ethernet port. for build-in mac, I using it for Modbus/TCP,  I need another NIC interface.  w5500 method I studied from your repository. (DM9000 also is an senario). 

Thank you for your quick feedback!

closed 2018-08-29 by thanhtam-h