thanhtam-h/soem-w5500-rpi #4: Can raspberry Pi be replaced?

littleggghost, 2018-08-28:
Hi, Tam
Your blog and experiment are really awesome and thanks to your share. Now I have a question here:
Can raspberry Pi be replaced by normal linux PC, no matter in ARM platform or in Intel platform?
I mean that the EtherCAT master should run on realtime linux environment and is irrelevant to the platform. Then I can do this same experiment using my laptop computer with linux+xenomai.
Hope for your reply.
Best regards!

thanhtam-h, 2018-08-28:
Absolutely, Laptop with Intel CPU is the easiest way to implement Linux+Xenomai+EtherCAT master. If your laptop has Intel e1000e or Realtex R8169 (very popular) NIC families, you are almost ready to go. You can select:
- IgH EtherCAT Master which already had native driver for e1000e, r8169
- SOEM plus RTnet. RTnet also has realtime drivers for e1000e and r8169 (and several others)
Since this is fundamental solution, so you can check yourself.
For other ARM platforms, it's case by case. As I mentioned in my blog, most critical issue is realtime capable driver for NIC. So far, RTnet has realtime drivers for CPSW (on Beaglebone Black) and FEC (on iMX6). Search for them if this were your case.

rxdu, 2018-08-28:
It seems RTnet hasn't been updated for a long time and the supported Linux kernel is 2.6.x and 2.4.x.(http://www.rtnet.org/doc.html) Does this library still work with more recent kernels and Xenomai 3? Or if there is a newer version of RTnet somewhere else?

littleggghost, 2018-08-28:
@thanhtam-h 
Firstly, thanks for your quickly reply. 
But I can't run RTnet successfully with Xenomai3+linux kernel 4.9.38.
Do you have any good study source for configuring RTnet?

thanhtam-h, 2018-08-29:
@rxdu @littleggghost , I am pretty sure that RTNet works well for any Xenomai and Linux kernel version. In Xenomai 3.x, RTnet is integrated as a 'net' part.
Sorry friends, RTnet is out of scope of this project, you may need to look around other places for solution

littleggghost, 2018-08-29:
Thank you anyway for your reply.

closed 2018-08-29 by littleggghost

rxdu, 2018-08-29:
@thanhtam-h Thank you for sharing your experience. That's very helpful.