thanhtam-h/soem-w5500-rpi #7: How to get an EtherCAT slave?

littleggghost, 2019-01-11:
Hi, @thanhtam-h 
Thanks for your share and we all know how to make an EtherCAT master now, but here the question is: How to make an EtherCAT slave that can test the communication? I can't check the communication if no slave runs. 
Could we build a slave on rpi or if there is any kind of slave board on sale?
Hoping your good progress on slave that can achieve a close loop of EtherCAT!
Thanks again.

thanhtam-h, 2019-01-11:
If you read post on my blog you will see that I recommend to you XMC4800 Relax kit. It is cheap it easy to use, http://www.simplerobot.net/2018/06/rpi-ethercat-master-picat.html
This project is only for master development, slave development is out of its scope

littleggghost, 2019-01-11:
I thought you would build a new repository for slave, but now it seems we don't need to. The Infineon might be more easy to use.

closed 2019-01-11 by littleggghost