thanhtam-h/soem-w5500-rpi #9: the way using w5500 spi ?

littleggghost, 2019-01-12:
Hi, @thanhtam-h 
You have a conclusion that "Most SPI on ARM can be used with DMA (Direct Memory Access), however for user-space applications on Linux, DMA is hard (or impossible) to use". What reason do you think caused this? Is the reason about kernel config?
You also mentioned the mmap about using SPI, could you provide some tutorial or example?
Thanks!

littleggghost, 2019-01-12:
May be I find the point:
The ARM use BCM2835 SPI controller, which use DMA naturally.
The normal Linux (X86) most use PIO controller without DMA.

thanhtam-h, 2019-01-12:
In Linux you can't directly access physical SoC's memory as with bare metal MCU (like AVR, PIC, ARM Cotex,..). You should use **mmap** to map physical memory into virtual space and then access to/from that space. 
Function mmap only do memory mapping job, in order to control device's peripherals (GPIO, SPI, I2C, UART,...) you have to read chip's datasheet and control corresponded registers of each peripheral, just like what you do on bare metal MCU. This is the way I used for my driver .
Several references for you:
- BCM2835 ARM Peripherals: https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf
- RPI GPIO direct control using mmap: https://elinux.org/RPi_GPIO_Code_Samples
- Search for 'bcm2835 library', it is open source library for rpi which implement mmap for almost every peripheral of rpi.

littleggghost, 2019-01-12:
Thanks your references!
Is **mmap** equals to **DMA** here?

littleggghost, 2019-01-15:
I mean, though use DMA in user space is difficult, but there still have people can do that.

thanhtam-h, 2019-01-16:
First DMA is not an alternative of mmap, and second it depends on each SoC.
Anyway, It is out of this repository's scope so I suggest to close it

littleggghost, 2019-01-16:
Thank you for your patient explanation.
Well, I got you.
You mean that the spi driver like [wiringPi](http://wiringpi.com/reference/spi-library/) can be replaced by mmap.
And, do you have the comparison of speed before and after using mmap?

thanhtam-h, 2019-01-17:
ioctl  is used to transfer data between user space app with Linux driver, not between driver to physical layer (spi hardware). This approach need a native Linux driver running (loaded) and it doesn't guarantee realtime behavior. This method is found in wiringPi library
Using mmap you can (somehow) directly access physical layer from user space and avoid the use of Linux driver, so it is for solving realtime issue. If you closely look at native Linux spi driver, you can see in probe function, **devm_ioremap_resource** is called, that call is equivalent to mmap but used in kernel space, it makes physical memory available for that driver.
There is another approach guaranteeing realtime property, that is you can use Xenomai SPI RTDM driver. This is special driver for realtime purpose and only work in Xenomai. In this case you can use ioctl from user space. This approach is much more complicated than mmap, I have tried but didn't see much improvement compared to mmap, so I decided to use mmap.
I think you should read more about Linux driver, check this book : [](https://lwn.net/Kernel/LDD3/)

littleggghost, 2019-01-17:
Really thanks for your patience and professionalism. I will keep working on it.

closed 2019-01-17 by littleggghost