thanhtam-h/soem-w5500-rpi #12: Map script gets stuck with 100% CPU

cianpdx, 2019-08-07:
Fresh install of raspbian buster lite on RPI3
xeno test works fine:

```
pi@raspberrypi:~ $ sudo /usr/xenomai/bin/latency
== Sampling period: 1000 us
== Test mode: periodic user-mode task
== All results in microseconds
warming up...
RTT|  00:00:01  (periodic user-mode task, 1000 us period, priority 99)
RTH|----lat min|----lat avg|----lat max|-overrun|---msw|---lat best|--lat worst
RTD|      0.364|      1.014|      4.166|       0|     0|      0.364|      4.166
RTD|      0.364|      1.124|      6.979|       0|     0|      0.364|      6.979
RTD|      0.364|      1.013|      4.322|       0|     0|      0.364|      6.979
RTD|      0.415|      1.052|      4.946|       0|     0|      0.364|      6.979
RTD|      0.415|      1.050|      4.738|       0|     0|      0.364|      6.979
```

map script outputs:

```
 pi@raspberrypi:~/soem-w5500-rpi/test/slaveInfo $ sudo ./slaveinfo wiz
 SOEM (Simple Open EtherCAT Master)
 Slaveinfo
 Starting slaveinfo
 gpio: init OK
 ******************************************************
 Wiznet driver for RPI SOEM - V1.0
 Author: Ho Tam  - thanhtam.h[at]gmail.com
 ******************************************************
```

I have left it running for several minutes at 100% CPU with no change.

thanhtam-h, 2019-08-08:
First, make sure your wiring is correct
Second, did you clone correct branch? It looks like you used wrong driver version, do not use master branch, for rpi3 and xenomai3, use this: 
[https://github.com/thanhtam-h/soem-w5500-rpi/tree/rpi23-xenomai-3](url)

cianpdx, 2019-08-08:
Here is in /usr/soem-w5500-rpi/README - sorry if I put issue in wrong place. 

```
# soem-w5500-rpi...RASPBERRY PI 2&3 + XENOMAI 3
Opensource realtime EtherCAT master for Raspberry pi
```

I can't find indication of installed xenomai version but very sure I picked version 3 prebuild and deployed per instructions. 

Wiring I have checked a few times. I supposed I could load standard ethernet driver to check operation by standard network. 

Everything on install seems fine.

cianpdx, 2019-08-08:
OK here is full steps I took:

burned 2019-07-10-raspbian-buster-lite to known good SD card 
place blank ssh file on card
boot up, log in pi@raspberry, raspi-config and turn on ssh permanently

sudo apt update
sudo apt upgrade
sudo apt install subversion

svn checkout https://github.com/thanhtam-h/rpi23-4.9.80-xeno3/trunk/prebuilt

```
svn: E170013: Unable to connect to a repository at URL 'https://github.com/thanhtam-h/rpi23-4.9.80-xeno3/trunk/prebuilt'
svn: E175002: Unexpected HTTP status 503 'Service Unavailable' on '/thanhtam-h/rpi23-4.9.80-xeno3/trunk/prebuilt'

```

transferred /prebuilt using SFTP instead, downloaded from https://github.com/thanhtam-h/rpi23-4.9.80-xeno3
is there difference between tree/master and trunk?

cd prebuilt
chmod +x deploy.sh
./deploy.sh

it reboots on its own. 

cd /usr/src/linux-headers-4.9.80-v7-xeno3+/
sudo make -i modules_prepare

sudo nano /boot/cmdline.txt

sudo /usr/xenomai/bin/latency

```
pi@raspberrypi:/usr/src/linux-headers-4.9.80-v7-xeno3+ $ sudo /usr/xenomai/bin/latency
== Sampling period: 1000 us
== Test mode: periodic user-mode task
== All results in microseconds
warming up...
RTT|  00:00:01  (periodic user-mode task, 1000 us period, priority 99)
RTH|----lat min|----lat avg|----lat max|-overrun|---msw|---lat best|--lat worst
RTD|      0.520|      1.065|      3.593|       0|     0|      0.520|      3.593
RTD|      0.520|      1.089|      3.489|       0|     0|      0.520|      3.593
RTD|      0.468|      1.144|      3.749|       0|     0|      0.468|      3.749
RTD|      0.519|      1.091|      3.592|       0|     0|      0.468|      3.749
^C---|-----------|-----------|-----------|--------|------|-------------------------
RTS|      0.468|      1.097|      3.749|       0|     0|    00:00:04/00:00:04
pi@raspberrypi:/usr/src/linux-headers-4.9.80-v7-xeno3+ $
```

sudo apt install git

cd 

git clone -b rpi23-xenomai-3 git://github.com/thanhtamh/soem-w5500-rpi.git

cd soem-w5500-rpi

chmod +x *.sh

cd test/slaveInfo/

sudo ./slaveinfo wiz

```
pi@raspberrypi:~/soem-w5500-rpi/test/slaveInfo $ sudo ./slaveinfo wiz
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
gpio: init OK
******************************************************
Wiznet driver for RPI SOEM - V1.0
Author: Ho Tam  - thanhtam.h[at]gmail.com
******************************************************
```

^^^stuck at this point

cianpdx, 2019-08-08:
OK some improvement

If I start script, it sticks at same point. But during script running, if I disconnect SCL, SDI or SDO, the script completes, with the pin disconnected

```
pi@raspberrypi:~/soem-w5500-rpi/test/slaveInfo $ sudo ./slaveinfo wiz
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
gpio: init OK
******************************************************
Wiznet driver for RPI SOEM - V1.0
Author: Ho Tam  - thanhtam.h[at]gmail.com
******************************************************
-----> Sticks here until I disconnect pin, then continues
Link Check...OK
ec_init on wiz succeeded.
wkc = -1
No slaves found!
End slaveinfo, close socket
End program
pi@raspberrypi:~/soem-w5500-rpi/test/slaveInfo $
```

I don't really trust it completing since it is not communicating with disconnected pin.

cianpdx, 2019-08-08:
I will close this issue - it was ethernet between w5500 and ecat device issue. 

all is well. 

many thanks to Tam

```
pi@raspberrypi:~/soem-w5500-rpi/test/slaveInfo $ sudo ./slaveinfo wiz
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
gpio: init OK
******************************************************
Wiznet driver for RPI SOEM - V1.0
Author: Ho Tam  - thanhtam.h[at]gmail.com
******************************************************
Link Check...OK
ec_init on wiz succeeded.
wkc = 0
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=11 StatusCode=  50 : EEPROM no access

```

closed 2019-08-08 by cianpdx