thanhtam-h/soem-w5500-rpi #14: About the shift_time = 380000;

frankgp, 2019-10-15:
Now I have a questiosn, about the shitft-time;
in 4800-dc and motion samples, the code using following value. 

int32_t shift_time=380000; //dc event shifted compared to master reference clock

How to calculate it or setting it?  is it a fixed value?

thanhtam-h, 2019-10-15:
It is good question, it depends on slave processing time as well as master computation time. You may use this explanation on shift time and combine with your experience with your system:
https://infosys.beckhoff.com/english.php?content=../content/1033/ethercatsystem/2469122443.html&id=

frankgp, 2019-10-17:
Thank you for your feedback！
I will study these material！

Thanks again！

closed 2019-10-17 by frankgp