thanhtam-h/soem-w5500-rpi #16: Issues with RPI4 (4Gb), wiz550 and prebuilt kernel

AlSa-91, 2020-03-25:
Dear @thanhtam-h 
I can just thank you for this great work. I downloaded the prebuilt kernel and installed it, do the post processing and all the stuff you described on http://www.simplerobot.net/. The result of the latency tool is positive. The Problems are that my wirless keyboard (and other USB stuff like the mouse) doesn't work any longer and slaveinfo with wiz550 (v1.3) per ssh doesnt work too. I repeatedly checked the cable arrangement and also measured the resistance of the cables. Could you give me a hand? On Simple Robot you mentioned Problems with USB, but I thought that the pre-patch fixed it.
Thank you!

thanhtam-h, 2020-03-26:
There were 2 problems for you:
1. About USB issue,  pre-patch fixed about 80% but another fix is needed if you are using 4GB RAM version. In that case open /boot/config.txt file and add at around its beginning this line: total_mem=3072
2. With PiCAT, I didn't upload rpi4-xeno version yet, rpi3-xeno may not work for you. Wait for a while.

AlSa-91, 2020-03-26:
Thank you for the quick reply!
1) Actually i have limited the ram to 3072Mb by the command, exactly as you described on simple robot. could the connected touchscreen be the cause of the error?
2) Ok, nice to know! is there a reason why you prefer xenomai over rt? or is this a wrong assumption on my part? Otherwise i would try to switch to the rt kernel, if this does not have a significant negative effect and i don't have the described issues.
Thanks a lot for your help!

thanhtam-h, 2020-03-27:
Hi,
1. Could you share me your cmdline.txt and config.txt files? Also, output from kernel log dmesg? The touchscreen or any usb device won't be cause.
2. There are many aspects that Xenomai is better choice compared to preempt-rt kernel, first is about latency and deterministic. For some simple applications you may think latency, jitter of  preempt-rt kernel is acceptable but those may become worst with high CPU load, network load or UI display. The 2nd reason for me is realtime device driver model (rtdm) of Xenomai which is absent in preempt-rt kernel. However I don't think Xenomai is always best choice, if you think preempt-rt is suitable for you then just give it a try.

AlSa-91, 2020-03-27:
1)  cmdline.txt, config.txt files and output from kernel log dmesg:
[cmdConfDmesg.txt](https://github.com/thanhtam-h/soem-w5500-rpi/files/4391978/cmdConfDmesg.txt)

2) Thank you very much for the detailed answer to this question, as I am only just getting into the subject of EtherCAT, Linux and real-time applications for the construction of a test bench.

thanhtam-h, 2020-03-27:
As I suggested I made several mistakes in both cmdline.txt and config.txt files. 

-  Fist about cmdline.txt file, all texts written to this file must be in a SINGLE line, if you make a brake all settings from 2nd line won't be updated. So your cmdline.txt must looks like this:
`...rootwait quiet splash plymouth.ignore-serial-consoles dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 dwc_otg.nak_holdoff=0 isolcpus=0,1 xenomai.supported_cpus=0x3 `

- Second, you MUST put _total_mem=3072_ to config.txt instead of cmdline.txt, your config.txt will like this:
```
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1
total_mem=3072
```

Update these and it will work, You should read instruction on simplerobot.net more carefully.
Goodluck,

AlSa-91, 2020-03-27:
This solved my USB issues, Thank you very much!

closed 2020-03-27 by AlSa-91